Jaffe-/NESizer2 #3: I Want To Build One

davidpgil, 2019-07-18:
I want to invest the time and resources needed to build one or a few of these. I don't have a background in electronic/electrical engineering, but I am willing to learn. Are the current specs and schematics complete enough that I should be able to build a NESizer if I followed them?

Joshkuebler, 2019-10-23:
I too am interested. Most of my experience is with hardware and at the component level but am learning c++ (school) and am gearing up to build something either like this, or this exactly. I hope everything is correct! Looks very promising. Any luck davidpgil?

davidpgil, 2019-10-24:
@Joshkuebler I have not attempted and I probably will not until I can get confirmation that this design would actually work. I don't have the background to determine if this is a working design.

Jaffe-, 2019-10-24:
Hi to both of you, and sorry for the late reply @davidpgil !

The hardware should work as it is shown. However, the last time I built a board was almost 4 years ago, so I can't remember all the details of how that build went. But as far as I can recall, the board I built worked without having to do any changes to it (patching with wires etc.)

There is one other person who has built the NESizer as well. He did get it up and running without many issues.

davidpgil, 2019-10-26:
@Jaffe- thanks for answering. So, since I have no experience with building something like this, and you dont remember details on how to build one, I think I should start an entry level project such as from [Make Analog Synthesizers](https://www.amazon.com/Make-Analog-Synthesizers-Electronic-Synth-DIY/dp/1449345220/) then try to build yours?

Jaffe-, 2019-10-28:
@davidpgil Yes, definitely. If you don't know that much about electronics then it is a good idea to do something that is simpler and is well documented first! Building the NESizer will require you to get a board produced, and to buy all the components that are listed in the Bill of Materials (listed in hardware/BOM.xls). It is quite a costly investment if you're not completely sure that you can do it.

Joshkuebler, 2020-01-04:
Davidpgil you should hit me up I just recieved the PCBs for this I had to order 5 qty. josh_kuebler@yahoo(dot)com