Jaffe-/NESizer2 #4: Need Help With Code!PLZ!

Joshkuebler, 2020-02-22:
Hey Jaffe, sorry to bother you, ive got the boards all put together, hardware and all, minus the coin cell battery(need to run to store). But i am a relative noob when it comes to compiling code. I have atmel studio installed. I see that on github there are many file/folders for all the various functions. Can you explain to me how to build code and load onto chip. I have an ftdi usb to serial will that work also i have an arduino uno if i can use that to program. Please any help would be greatly appreciated, i am so close to finishing this beautiful build. Ive dreamed of putting together an nes synth for so long! you can email me josh_kuebler@yahoodotcom

Jaffe-, 2020-02-23:
Hi again Josh.
I've attached the latest NESizer firmware code (built from github).

To build this on Windows, I first installed git for Windows and cloned the
repository. Then, using the git bash shell I ran 'make' from the cloned
repository. That should work out of the box if you have avr-gcc installed.

Regards,
Johan

Joshkuebler, 2020-02-24:
Johan you are a stand up guy. Thanks a million. I will be sure to send video once I get it all sorted. Have a great week!-Joshua Kuebler