Jaffe-/NESizer2 #5: Documentation sample code does not compile

ryancole38, 2021-01-26:
Hi! First of all, I wanted to say thank you for the tremendous work you've done and the effort you've put into documenting the project.  I've been working on reverse-engineering the project to make a cut-down version (mostly no RAM or front panel) and your documentation and comments have made this task much easier than it would have been otherwise.

That said, the sample code in `software.md` does not currently compile.  I believe it alludes to an older version of the code that had separate functions for each channel.  From my understanding, I believe that this is the minimum amount of code required to produce a noise:

```
main() {
        bus_setup();
	io_setup();
        apu_setup();
    
	sq_setup(0, &sq1);

	sq1.period = 400;
	sq1.duty = 2;
	sq1.volume = 15;

	sq_update(&sq1);

	apu_refresh_channel(CHN_SQ1);

	while(1){}
}
```

I somehow killed the 2A03 that I was working with (oops), so I can't confirm that this does produce a sound.  In my mind, `bus_setup()` is required because it sets up the bus pins that allow writing to the register that feeds the 2A03.  Similarly, I believe `apu_setup()` is required because it enables the outputs of the channels.

Also, I had to change the fuses on the ATMEGA328P in order to get it to work off of the same clock as the 2A03.  I'm not sure if I selected the right clock divider setting, but I think this would be helpful to add to `software.md`.

If these changes make sense to you, I would be more than happy to update the documentation.  Once again, thanks for your work!

KeenVox, 2021-09-13:
Hi, 
what fuses need to be changed on the atmega? I compiled the firmware but I'm not sure if I need to set the fuses too before programming.

Jaffe-, 2021-09-15:
Hi,

The fuse values should be: E:FF, H:DF, L:E0

Johan

KeenVox, 2021-09-15:
Hi, I asked for the fuses but deleted the comment to start a own issue. Sorry for the confusion  🙂
But now I don't need. 
Just received the display and looking forward to assemble the synth!