Jaffe-/NESizer2 #6: How to upload samples?

KeenVox, 2021-09-15:
Hi,
I just wanted to ask, how to upload samples? I assume the samples have to be converted for sysex and then send while in sample select mode?

Thanks in advance!
Keen

KeenVox, 2021-09-15:
Ah, okay. I will try it!

Thanks for you help, and for the fuse settings too. I'm relativly new to programming microcontroller outside of arduino, so I really appriciated the support!

Jaffe-, 2023-10-06:
Hi @KeenVox,

Really late follow-up, but I recently started working some more on this project to try and finish most of the features.

If you're still interested, I have now done some fixes to the code such that uploading and using samples is a bit more user friendly. I actually realized that the LED progress bar feature was actually never pushed to the git repository, but I have done that now.

I have also added a tool for generating a sysex file with the correct format.
To use this tool, you can look at the script convert-sample.sh. To put it short, convert the sample file to 16 kHz, 7-bit. unsigned raw format. SoX is a good utility for this, the script shows how to do this conversion for input files in wav format. 

To create a sysex file from the converted raw sample, run the gensysex program like this:

`./gensysex sample <index> <input raw sample file path> <sysex filename>`

Where the index is which index you want the sample to have in the NESizer file system. Index 0 corresponds to MIDI note 60 (middle C). This is the note you will have to input on MIDI to trigger the sample on the DMC channel, or enter into the sequencer if making a pattern with DMC samples.

When you send the sysex data from the file, the NESizer should give you a progress bar on the LEDs, and the two digit display should show 'rr' (for now). If you see this, you can be sure that the NESizer is actually receiving the data and is storing it in memory. If it does not start, I would try to format the sample file system first, which is done by going into the settings mode/page and then pressing the 7th button on the second row.

beau-seidon, 2023-10-06:
Hell yes! I know what I'm doing this weekend :). I have been meaning to experiment with this to get some samples in, but have been super busy with work lately. I'm going to try it out with your script.

Probably better to post this on another thread since unrelated to samples, but over the next month or so I will try to add some MIDI CC callbacks to modify some of the voice parameters like duty and the envelopes etc., to match the Famimimidi MIDI implementation. It takes me a while because I'm still learning C/C++.

I'll finish the build guide from my notes first, like I said I'd do previously. It's about 40% done.

Jaffe-, 2023-10-08:
> Hell yes! I know what I'm doing this weekend :). I have been meaning to experiment with this to get some samples in, but have been super busy with work lately. I'm going to try it out with your [...]
That would be really great! I'd be happy to review a pull-request and get this in.
> 
> I'll finish the build guide from my notes first, like I said I'd do previously. It's about 40% done.
Really great work of you to make a build guide and updated BOM. I also would like to get the gerber files uploaded so people can order this without having the PCB tool. Don't know why I didn't do that already...

KeenVox, 2023-10-10:
Hi Johan,
very cool! Will definitly try everything out. Great to see you working on
this project again!

Jaffe-, 2023-10-10:
> Probably better to post this on another thread since unrelated to samples, but over the next month or so I will try to add some MIDI CC callbacks to modify some of the voice parameters like duty and [...]

Regarding another thread for this, please make one under Discussion (instead of Issues), I added that feature to the repository now :)
Also, I pushed a commit yesterday which changes the Makefile to compile the sources with -O2 instead of -O3 optimization level. This reduced the flash size from ~30 kB to ~22 kB (less aggressive unrolling of loops and branches), so there should be enough room to implement more features now.
I did some quick testing that things still run fast enough, and it seems like it is.

jeff464, 2024-08-13:
I would also like to know how to upload samples to the NESizer. I converted my audio file to the proper format using sox, but I'm not sure what "run the gensysex program like this: ./gensysex sample <index> <input raw sample file path> <sysex filename>" means in this context. The windows command line just errors out and double clicking the file sends me to VSC to edit it.

beau-seidon, 2024-08-14:
the source for gensysex is in the repo, and can be compiled with gcc or another c compiler (i used msvc). ill upload my Windows compiled binary after work. im also working on a script to batch-upload sysex files to nesizer using midiox, to load whole drum/sample kits at a time.

beau-seidon, 2024-08-15:
[gensysex.zip](https://github.com/user-attachments/files/16625390/gensysex.zip)

here are the gensysex and deltacompress windows executables used to convert samples. you shouldn't need to do anything with dc. 

take a look at the .sh script in tools to see how the process is automated in linux. it shows the syntax for calling gensysex in a terminal. I have a .sh and .bat script for converting big batches of samples, but I have some tweaks to make before I upload them. 

I also have a little bit of experimenting to do regarding the indexing of samples (mapping to MIDI notes), which may lead to a wishlist item of my own.

jeff464, 2025-01-31:
I'm about five months late on this. Sorry!

I just followed the instructions in the manual. Using sox, I am able to generate a new sample named "sample.raw". Perfect!

However, when I place "sample.raw" in the same folder as gensysex and run gensysex.exe via a cmd prompt in the manual using "gensysex sample 8bit 2 sample.raw sample.syx" I receive a "Failed to open input file 2: No such file or directory" error.

I'm not sure what I'm doing wrong here.

beau-seidon, 2025-01-31:
try it without 8bit

jeff464, 2025-01-31:
I just tried it without 8bit and it successfully generates a sysex file. When I send the sysex to the NESizer, the segment display changes to something unintelligible. When the transfer is finished, nothing seems to have loaded.

beau-seidon, 2025-01-31:
what software are you using to send the sysex file?

jeff464, 2025-01-31:
I tested with Borne SendSX 1.40 and MIDI-OX

beau-seidon, 2025-01-31:
I use MIDI-OX with the following settings: 

[image]

Use `Command Window > Load File` to open the sysex file you generated, and look at the first few bytes. They should be **F0 7D 4E 02** if you chose index 2. Is that the case?

Are you using the latest firmware build?

jeff464, 2025-01-31:
Yep, using the latest build. First few bytes are F0 01 02 00 1F, so it seems that something is going sideways.

beau-seidon, 2025-01-31:
Are you using gensysex.exe from the /Tools folder? or the old one in the linked zip file from earlier in this thread which im about to remove?

jeff464, 2025-01-31:
I started with the one in the tools folder before I knew to remove 8bit and moved to the one in this thread. I'll try the other again to see if I have any luck.

beau-seidon, 2025-01-31:
I'd like to know what exactly happened when you tried to load the syx file with the incorrect header. It should have printed "no" to the 7seg display until it finished sending.

jeff464, 2025-01-31:
Trying to run the gensysex.exe file from the main tools download with/without admin - a double click gives no response and manually running it in cmd displays "'gensysex.exe' is not recognized as an internal or external command,
operable program or batch file."

jeff464, 2025-01-31:
[image]

Here is what the display shows when trying to upload the sysex file:

jeff464, 2025-01-31:
[image]

Here is what my display shows when trying to upload the sysex file.

beau-seidon, 2025-01-31:
It's a command line tool, so you have to use the terminal.
Make sure your current working directory is the folder gensysex is saved to. 
Use it exactly like you did the other one.

beau-seidon, 2025-01-31:
> [image]
> 
> Here is what my display shows when trying to upload the sysex file.

oh thats weird. mine shows those two top horizontal lines in the middle segments. it's supposed to say "no" in lowercase letters.
are there any other display issues? any garbled symbols when you expect numbers?

jeff464, 2025-01-31:
Nope! Everything is perfect as far as I can tell. Thinking about it, I think it did say "no" the very first time I tried. Now it displays what's in the photo.

beau-seidon, 2025-01-31:
i'll see if i can break mine by feeding it an old-format sysex!

jeff464, 2025-01-31:
Good news on both fronts - a reboot fixed the weird display. It now displays "no" correctly when sending it garbage. Also, I successfully uploaded a clap lol. It seems my problem was two fold - I was using the wrong gensysex, and I needed to remove "8bit."

Now to stay up all night and dump all sorts of stuff to the NESizer. Thank you for talking me through this.

beau-seidon, 2025-01-31:
YAS! load all the samples! and no worries. 
sorry i didn't catch the command error in the docs. will get that updated.

bad news is you still have to upload samples to the board one at a time if you're using MIDI-OX. 
I was able to write COM scripts in 2 different languages (ps, and vbscript) to automate MIDI-OX and upload samples in batches. unfortunately theres a bug in MIDI-OX that makes it quit randomly halfway through. 
i thought i reported it on their forum months ago, but looking back through them tonight i don't see it, so i'll go type something up, but im not sure anything will result from it. if it was open source i'd try to fix it myself.

jeff464, 2025-01-31:
Bummer! Are there open alternatives? That would be an enormous help, but truth be told, I'm just happy it's working now. If I need to spend a little extra time uploading one sample at a time, that's totally fine!

beau-seidon, 2025-01-31:
try Bome Send SX again, and see if it lets you select multiple files to send in batch. I have never used it before. 
If not, there's lots of open source midi libraries, i can slap a script together in python or something soon and see what happens.

beau-seidon, 2025-01-31:
if you're a linux guy, there are instructions for using the `amidi -p <MIDI device> -s <sysex file path>` command to upload samples, and you could easily google up how to write a for-loop in a bash script to batch upload.

jeff464, 2025-01-31:
I'm not too knowledgeable when it comes to Linux, unfortunately. It's on my bucket list, and this might push me to jump in. As far as Borne goes, it doesn't look like there's a way to send bulk files. No biggie!

beau-seidon, 2025-01-31:
well now i want to use wxWidgets or something to make a simple cross platform GUI application to convert and upload samples, using the current resources like sox, and gensysex, and whatever i else i come up with to send the sysex files. 
that's going on my to-do list! not super close to the top, but it's on there now...