Jaffe-/NESizer2 #7: Nesizer goes crazy when cell battery is inserted

KeenVox, 2021-09-24:
Hi,

I build the NESizer2 and everything works great so far, as long as I don't insert the battery for patch saving.
When I power on the NESizer with the battery inserted the machine goes crazy. All soundwaves are activated and it starts to play a random melody while showing random numbers on the display. Without battery everthing works fine again. 

Any suggestions for troubleshooting?

KeenVox, 2021-09-24:
Seems to be a ram problem. When I remove the left chip, anything works like it should...

Jaffe-, 2021-09-24:
I think the RAM chip is probably OK, but the contents are just random.
First time you use the nesizer with new battery you should also do a memory reset to format the storage system. 
I would do these:

PATCH_RESET (button 8 - first button on second row counting from the left)
PATTERN_FORMAT (button 9 - second button on second row counting from the left)
INIT_SETTINGS (button 11 )
SAMPLE_FORMAT (button 14)

(Of course there should be one single option to do all these at once...)

Do this with both RAM chips installed. When you have done it, turn power off and on again and it should hopefully work!

KeenVox, 2021-09-24:
Thanks for your reply. That helped. It's a bit sad, the ram melody was quiet nice. :-)

When I reseted the settings (11) and formated the samples (14) the Nesizer crashed and freezed, could this be because there were no settings or samples loaded onto it? 
Beside this everything works fine.

Jaffe-, 2021-09-24:
It should not crash after you reboot it? Doing the various formatting operations can take some time though, and the UI will be frozen until it is done.

KeenVox, 2021-09-24:
Ah, okay. Maybe I was too fast to turn it off when I pressed SAMPLE_FORMAT. The first two actions (8,9) where so fast, so I thought the system crashed. 
Will try it again.

KeenVox, 2021-09-24:
> It should not crash after you reboot it? Doing the various formatting operations can take some time though, and the UI will be frozen until it is done.

Everything seems to work like it should now. But I have on little problem left. When I start the synth, there's a hanging note on the noise channel. I have to play a note on the noise channel once to make it stop, after that, everthying is fine.
I'm using a famicon clone chip, could that be the cause here?

Jaffe-, 2021-09-24:
I remember this being an issue. If you find a fix for it, feel free to make a pull request! :) 
And congratulations for building what I would think is the second working nesizer!

KeenVox, 2021-09-25:
Ah, ok. Good to know. If I ever get into coding, I will try my best. 😆
Thanks for the kind words. I don't get why your nesizer didn't get more recognized. It's a really nice and complex synth  🙂.
The only problem for others may only be to source some parts, sadly.

KeenVox, 2021-09-25:
[image]

beau-seidon, 2023-06-22:
> Thanks for your reply. That helped. It's a bit sad, the ram melody was quiet nice. :-)
> 
> When I reseted the settings (11) and formated the samples (14) the Nesizer crashed and freezed, could this be because there were no settings or samples loaded onto it? Beside this everything works fine.

When I booted for the first time, I experienced the same noisy looping sequence, and since I read this thread before I built it, I decided to record a sample of it before formatting the RAM. 
If you would like a copy the recording for your own use, I would be happy to send it to you.

beau-seidon, 2023-06-23:
> Everything seems to work like it should now. But I have on little problem left. When I start the synth, there's a hanging note on the noise channel. I have to play a note on the noise channel once [...]

> I remember this being an issue. If you find a fix for it, feel free to make a pull request! :) And congratulations for building what I would think is the second working nesizer!

I reckon it might stop the stuck noise and triangle notes after a reboot if you called some "channel_initializer" function just once after everything else starts up. 
It might simply wrap the calls to `assigner_notify_note_on(midi_channel, 60)` and then `assigner_notify_note_off(midi_channel, 60)` on all MIDI channels (or just the MIDI channels assigned to those oscillators, since they are stored) with data1 as Middle C or maybe something less audible.

Since playing notes un-sticks the channels, and those are the called functions when notes are played, just call 'em right away to un-stick them preemptively.

It's more of a workaround than a proper fix, but I'll see if I can make it work locally on mine and report how it goes.

KeenVox, 2023-06-30:
That sounds awesome,. If your modified code works, I would love to get the hex, to update mine. The sticky channel really gets on you nerve after some boot ups. ;)

beau-seidon, 2023-07-10:
This .hex works to fix the hung noise/triangle note at boot, with the following requirements:

1. Both the triangle channel and the noise channel must be assigned to any MIDI Channel 1-16. The default is 0 after flashing, so go change them. They don't have to be assigned to the same channel.

2. The triangle channel and noise channel must both be enabled at startup. I found that saving preset 00 with default settings and all channels ON works perfectly.

[nesizer_fw_startuphack.zip](https://github.com/Jaffe-/NESizer2/files/11997884/nesizer_fw_startuphack.zip)