Jaffe-/NESizer2 #8: Unable to compile firmware properly with avr-gcc

SonicJr623, 2021-12-06:
Hello Jaffe. 

Sorry to bother you.
I would like to build my own NESizer, but I am unable to compile the firmware properly.

Here are the exact steps I tried to compile.
I first installed avr-gcc and git for Windows and cloned the repository.
Then I run 'make' from the cloned repository using the git bash shell.
However, I get the following error message.

In file included from main.c:29:
apu/apu.h:29:10: fatal error: sample/sample.h: No such file or directory
   29 | #include "sample/sample.h"
        |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:41: main.o] Error 1

Could you please help me with some advice? Thank you very much!

Best regards
Sonic

Jaffe-, 2021-12-06:
Hi Sonic,

This looks like a fairly simple compilation issue where the include path given to gcc is wrong. This is done in line 15 of Makefile, with the option `-I$(shell pwd)`. I suspect the "shell pwd" part might not work in your case. You can try to replace this with a hard coded path to where the code is, something like `-I C:\...`

Another option, which I would recommend is to install Windows Subsystem for Linux (WSL), which would provide a better environment to build the code in.

SonicJr623, 2021-12-07:
> 
Hi Jaffe,

I tried hard-coding the code path into the Makefile and now I can compile the firmware properly, thanks a lot!

Best regards
Sonic

closed 2021-12-07 by SonicJr623