Jaffe-/NESizer2 #9: How can I compile the sources

cgerrist, 2022-06-10:
HI,

with with compiler do you compile the project.
There is no Project description in the src folder.
I try code from microsoft but it did not compile without errors.

Chris

Jaffe-, 2022-06-12:
Hi Chris,
The code has to be compiled for the Atmega328 with avr-gcc. On Linux it should be enough to install avr-gcc and avr-libc,  and Make. It can then be compiled by running make.
I have not tried to compile it on Windows. I think the easiest way to do it would be to install WSL (Windows Subsystem for Linux) and follow the previous description for Linux.
I guess it can also be compiled by installing Atmel Studio.

cgerrist, 2022-06-12:
Thanks for the reply. I live in Holland and we have only the PAL version of the sound chip. This chip is different the RP2A07. I am using a Mac. Tonight I wil try to compile it again.

Jaffe-, 2022-06-12:
On Mac it should be possible to install avr-gcc as well. I found this, for example: https://bytes.usc.edu/ee109/macos-64-bit-toolchain/
Seems like you can get avr-gcc from installing the Arduino IDE (some Arduinos use the same chip and the IDE uses avr-gcc to compile the arduino code).

You also need the avrdude tool to actually flash the Atmega chip with the compiled firmware.

Regarding RP2A07 it should work fine, the atmega firmware will determine which version is installed and do the necessary compensation for PAL vs NTSC clock speeds.

cgerrist, 2022-06-12:
Thank, I was reading about this chip and is a combine with the 6502 processor. When I had a Apple II I was programming in assembler for the 6502. So I got warm feelings for this chip 😀. This programming was a long time ago. When I was 14. And now I am 58.

cgerrist, 2022-06-13:
Hi, I was able to compile the sources. output: sesizer_fw.elf, hex and map

This was the easy step I think :-)

Thanks for the help

Jaffe-, 2022-06-13:
> Thank, I was reading about this chip and is a combine with the 6502 processor. When I had a Apple II I was programming in assembler for the 6502. So I got warm feelings for this chip grinning. This programming was a long time ago. When I was 14. And now I am 58.

It should be familiar to you then, you can see in the sources how the 6502 in the 2A07 chip is 'tricked' into doing what we want (load audio synthesizer registers) by just force-feeding it instructions from the Atmega microcontroller :)

> Hi, I was able to compile the sources. output: sesizer_fw.elf, hex and map
> 
> This was the easy step I think :-)
> 
> Thanks for the help

That's great!

cgerrist, 2022-06-13:
> It should be familiar to you then, you can see in the sources how the 6502 in the 2A07 chip is 'tricked' into doing what we want (load audio synthesizer registers) by just force-feeding it instructions from the Atmega microcontroller :)

I did not read all the sources yet so I did not see this part of the code. Is this all Atmega code or do you have a part logic that you dumb in the memory of the 6502 and a separate part ATmega?

Jaffe-, 2022-06-13:
The 6502 does not have any memory connected, the address bus is just unconnected. The Atmega is just giving it instructions on the data bus and waits for the appropriate number of clock cycles before putting a new one on the bus. It is described in more detail in https://github.com/Jaffe-/NESizer2/blob/master/docs/software.md

The first Nesizer design was based on only the 2A03/2A07 but it turned out to be quite limited how flexible the resulting synthesizer would be as an instrument. The Atmega is quite a bit faster and can do more of the extra computations needed for LFOs, envelopes, sequencing, MIDI etc.
I did consider to share the burden between the two by putting some code in a RAM or ROM for the 6502 to run and make a sort of command interface between the Atmega and the 6502, but when I found the idea of just spoon-feeding the 6502 in another project, I went with that solution. Running code in the 6502 would require more components and be a more complex design.

cgerrist, 2022-06-13:
I look to the layout of the NES, I think it is very easy to connect memory to the CPU. The NES has one 128x8 DRAM.
The NeSizer use 2 x 512x8 chips. Why do you need this? It is not connected to the adresbus

cgerrist, 2022-06-14:
Good morning. I did read hardware.md and looked at the layout I do not understand how the 2 x 512 K eat is connected to the address latch. Also How will the 2A03 make connection to the memory. I looked at the NES schema and is it not easier to use the RAM from the NES. Maybe stupid questions. But I would like to know how everting works before I build The NESizer. Did you order the PCB or made it yourself?

Chris

Jaffe-, 2022-06-14:
> But I would like to know how everting works before I build The NESizer. In addition, did you order the PCB or made it yourself?
> 
> Chris

I ordered PCB from a manufacturing service called Elecrow. Don't know if they are operational any more, but there are other similar services that manufacture PCBs quite cheaply these days.

cgerrist, 2022-06-14:
Normally I use my cnc to mill the pcb. Maybe I do this also with this project. Thanks for the clarification.