Jaffe-/NESizer2 #10: BOM Inquiry

beau-seidon, 2023-05-26:
First, this project is amazing, and thank you for sharing it. 

I'm prepping to build one right now, and I just sent the 2 PCB's to be manufactured (with a red upper and blue lower board like Super Mario's clothes, see attached pic!).

My question is this: in the BOM, IC13 component DS1210 has no quantity, is it no longer required? 
I noticed that in the last hardware.md commit (8 years ago, wow) you removed references to DS1210.
I compared the before and after, and see that a transistor switch circuit keeps the SRAM powered now instead, but DS1210 is still on the BOM. Is this just for reference / historical purposes?

It is also discontinued from Mouser but there is an equivalent replacement. If I don't need it, I'll save 10 bucks.

[image]

Jaffe-, 2023-05-27:
Hi,

Happy to share the project, and always cool when someone wants to make it! I know about two people who have tried to build it before, and at least one of them got it working (see other github issues) - so it is possible! :) 

To look at the history of the hardware design you should look at the history for the schematic and board files. I didn't always update the hardware.md file when I did changes to the board designs.

As I recall it, the DS1210 was out for a while because I thought I could design the battery backup circuit without using that component, but it turned out to be way too battery consuming. If I did it today I would use CMOS chips or FETs for that.
Anyway, I ended up putting the DS1210 back. The correct quantity should be 1 (as seen on the board). But if it is hard to get a hold of, it could be worthwhile to check if there are other compatible ICs for this. 

Please also take BOM.xls with a little grain of salt, I would definitely double check if there is anything missing in it. It should be pretty much in sync with the boards, but a resistor or two might be missing or have the wrong value. Definitely a good idea to have a cheap multi resistor kit when doing this project I think.

Love the PCB colors, good luck with the build!

beau-seidon, 2023-05-27:
Thanks for responding! 
I have read the other issues here and all the discussion I could find on Reddit and your YouTube videos, as well as some of the general info on NESDev, and looked over some of the IC datasheets, a bunch of other things. 
I'm trying to be as prepared as possible since I've only made a few Atmega based projects before, and none this size.

The DS1210 is marked obsolete and is replaced by MXD1210CPA+, and another with a different temperature tolerance.
The difference between new and the original is the operating supply current which was reduced from 5 mA to 230 uA.
Description of function, pinout, and recommended circuit from datasheets are the same.

Ok, so if the mainboard.sch and panel.sch files are reliable, I'll go part-by-part thru the BOM and diagrams and check them all and update the BOM if I find any differences. 
If the MXD1210CPA+ works and I get the whole thing running without melting something, I'll include that Mouser part number in the BOM as well, since it is well-stocked. 

I have a few of the listed things already. Lots of resistors and a some caps and a couple sockets and things. 
Was going to load the BOM into Mouser's template, and then subtract the parts I have.

beau-seidon, 2023-06-21:
It's alive! 

It works great. I'm going to upload an updated BOM that matches your latest .sch file with in-stock Mouser part numbers, then type up a guide with pics explaining how I ordered the boards, built it, and programmed it using an Arduino Uno (as an ISP). 
I'll write it at a n00b level, mostly because thats the extent of my capability, but also so nobody else has to struggle to figure any of it out like I did. Although that may have been the most fulfilling part.
 
I know I probably could have asked more for assistance, you have been super helpful and I gained much from reading your responses to others' questions, but I wanted to learn all that I could on my own first. 
But this is such a cool project, I want anyone to be able to build it worry free.

The only major issue so far is a firmware bug in midi.c, regarding how it handles MIDI messages comprised of a NoteOn status byte with Velocity=0 in data. 

Per the MIDI spec, any compliant synth should handle that command identically to a NoteOff message, but NESizer treats it like any other NoteOn command. 

I found this because Cakewalk (my favorite DAW) apparently stops MIDI notes using the NoteOn,Vel=0 method. 
Neither my hardware MIDI keyboard, nor Ableton Live, nor FL Studio had any issues playing sequences through the NESizer, but it spazzed when I fed it notes from Cakewalk. 
I experimentally passed the MIDI messages from Cakewalk through a loopback driver into MIDI-OX and found the difference in the status portion. 

I'm still learning C++ and have been super busy lately, so I haven't tried reprogramming it yet. 
In NESizer2/src/midi/midi.c, line 63 is the case MIDI_CMD_NOTE_ON inside the midi_channel_apply function. Thats where the bug resides. 
The following if statement should check the data byte when it gets a NoteOn command, to see if the velocity is 0, and when true just do the same thing as the MIDI_CMD_NOTE_OFF case. 
I'm going to hack at it when I have time to see if I can get it to recompile and play nice with Cakewalk.

Also, here are a few pics of my finished boards! 
I'm going to be recording some tunes with my new favorite toy in the future and I'll post a link to them here.

[image]
[image]
[image]

beau-seidon, 2023-06-22:
Programming with Arduino as ISP was very easy after watching a couple Youtube videos about avrdude and setting fuse bits. 
This one was particularly helpful: https://www.youtube.com/watch?v=Q2DakPocvfs

I used the Arduino Uno dev board rather the the RaspberryPi GPIO pins because the Pi doesn't tolerate 5 V on its pins, and the Atmega in the NESizer would be operating at 5 V. 
I read that it's fine to program one with a Pi, as long as the Atmega is supplied with 3.3 V instead of 5 V.

Arduino IDE comes with an example sketch to turn any Arduino board into an ISP programmer. 
After uploading that to my Uno, it was just a matter of matching the right pins with jumper wires. I did connect it directly to the ISP header on the board. 
I powered the NESizer using a 12 V DC (center-positive) power supply, and left the 5 V pins disconnected between the Arduino and the ISP. Arduino was powered by the USB cable connected to my laptop. 
I'll detail all this stuff a little bit more in my build guide

Tonight I slightly modified the midi.c, ui.c, and ui.h code then recompiled and everything seems to be working fine. 
My board responds correctly now to Cakewalk's MIDI note commands and I'm super excited about it!
I forked and modified and created a pull request on here too. It was my first time, I hope I did it right. 

As far as the user interface, I did notice that the silkscreen labels for some of the buttons on the PCB don't match the labels in the manual, but the manual seems to match what the buttons do when I poke them, so I have been using that. 
Might print some tiny stickers for the caps.

I wonder how difficult (if possible) it would be to use the popular Arduino MIDI Library by FortySevenEffects for expanding NESizer's MIDI capabilities. 
https://github.com/FortySevenEffects/arduino_midi_library

Jaffe-, 2023-06-22:
The labeling on the board is a bit out of sync with the current UI functionality. Some labels would definitely help! Actually a nice case with labeling on it would be the best. 

I took a look at your PR. It looks like it will fix the issue, but it also does a bit more than it needs (see the comments). Now you can try the process of fixing/updating a PR too :)

I'll take a look at that library, maybe it (or some of it) can be used. There isn't that much space left at the moment, but I know that there are some things that can be optimized in terms of space, especially with regards to the AVR standard library code that is linked in.

beau-seidon, 2023-06-23:
I gave the PR update a shot. My attempt was not elegant, but the code is fixed lol.

DannyyVoid, 2024-04-18:
Hi beau-seidon, based on your success to build this project I got inspired and start to get everything ready to build mine.
Did the current BOM is the one you used? I saw that you where planning to release an updated BOM
> It works great. I'm going to upload an updated BOM that matches your latest .sch file with in-stock Mouser part numbers, 

Anything I should know before ordering all the components from Mouser? I already replaced the DS1210
> The DS1210 is marked obsolete and is replaced by MXD1210CPA+

For PCB manufacturing, do you recommend any page in specific? I've used OSHpark long time ago.
Thanks in advance to both

beau-seidon, 2024-04-19:
> Did the current BOM is the one you used? I saw that you where planning to release an updated BOM

https://github.com/Jaffe-/NESizer2/issues/13#issuecomment-1695009960

I posted my mouser cart / BOM as a response to a different Issue thread. I haven't uploaded it "officially." For the most part it should be the same, theres a couple small differences though. Of the top of my head, the value of the bus resistor was different.

> For PCB manufacturing, do you recommend any page in specific? I've used OSHpark long time ago.

I used JLCPCB for mine and it was easy and everything worked perfectly. Took a couple weeks to arrive, but they keep a current status of the manufacturing process online and the delivery is tracked. I did consider using OSHpark, and I will probably use them for my next PCB based project. They will make single boards if thats all you want, and they're a cool purple and gold design. Minimum order from JLC was 5 boards each, but they have color options. There are many other manufacturers, but thats the limit of my experience.

DannyyVoid, 2024-04-19:
Thanks, I validated your BOM/Mouser Cart and everything still available, maybe in a couple months some capacitor might need change but that shouldn't be any problem. 
Right on! I'll order the parts and decide where to buy the PCB (Do you have any board available for sale? (since you said you had to order 5 of each) if not, no worries, I'll just order more online

beau-seidon, 2024-04-25:
Actually yes I still have the rest. I don't know how to PM on github but if you send me a delivery address and pay for shipping and a few bucks for the boards I'll send you a pair. Unless you ordered a batch already, I see it's been a few days since you asked. 

Note: mine are red for the interface panel and blue for the audio panel, not black. @circuitcreature built a black one somewhat recently. I don't know if they had singles printed or a batch with leftovers they might be willing to part with.

DannyyVoid, 2024-04-25:
I made an actual BOM based on the last revision @beau-seidon made, I only had to replace a capacitor that was EOL and no longer available. Hope is useful 
[NESizer2 BOM Rev2.xlsx](https://github.com/Jaffe-/NESizer2/files/15112995/NESizer2.BOM.Rev2.xlsx)

beau-seidon, 2024-04-26:
sweet! thank you. sent you an email about the pcb's

jeff464, 2024-06-09:
Hey everyone, I'm looking at the original BOM, the updated DannyyVoid BOM, and a part I ordered before I knew of the updated BOM. The original (obsolete) resistor network (mouser #858-L091C472LF) is rated at 4.7k, so I ordered a resistor network (mouser #652-4609H-1LF-4.7K) which is also rated at 4.7k, but I'm seeing that the new, updated BOM lists the new resistor network (mouser #652-4309R-1LF-15K) at 15k. Did I totally screw this up? Perhaps there was an update to the needed parts?

beau-seidon, 2024-06-09:
> Hey everyone, I'm looking at the original BOM, the updated DannyyVoid BOM, and a part I ordered before I knew of the updated BOM. The original (obsolete) resistor network (mouser #858-L091C472LF) is [...]

That component is a pull-down resistor for the switch input buffer / grid columns. At 15k it should pass about 0.33 mA to ground for each simultaneous button press. At 4.7k the current will be ~3x as much, but still only about 1.06 mA. Just don't push all 24 buttons at the same time, and it will probably be fine! I think only one row is ever turned on at a time, so the max current draw through the latch's button pins if you push all the buttons should be about 8.5 mA. The latch datasheet says output rating at 5 V is 6 mA, but the Absolute Maximum Rating says 35 mA. Not sure if that's the whole chip or a single output.

My guess is that in the worst case, you will have to replace the latch if you kill one or more of the three row output pins. If your buttons quit working after you set a really big book on top of your Nesizer while it is powered on, replace the latch and swap the bus resistor with a 15k.

Otherwise maybe watch out for less responsive buttons because more current will be going to ground so you'll have a bigger voltage drop in the button signal before it gets to the data bus, but I highly doubt it will be noticeable.

Someone with proper electronics training please correct me if I said something incorrect!

jeff464, 2024-06-09:
Got it, and that makes total sense. I was just initially going on the BOM that's included with the main project download since I wasn't aware of the updated BOM located here. Now I'll just update everything I haven't soldered yet. Easy enough.

That said, is there a reason the now outdated BOM hasn't been officially replaced with the new one in the main project download?

jeff464, 2024-06-09:
Also, there goes my plans on using my NESizer as a book shelf while sequencing!

jeff464, 2024-06-09:
Wait, just realizing there are now three BOMs - the original, the DannyVoid BOM, and the beau-seidon BOM (plus mine as a fourth that I threw together before I knew of the other two). Looks like my afternoon is now spoken for lol

circuitcreature, 2024-06-09:
I'm in the middle of rebuilding/building a new synth and started another BOM, so you can add that to your list. The one i'm working on includes LSCS and Digikey. I'd like to see yours as well, I'm finding deprecated parts, even on the list i've started a few moths ago.

jeff464, 2024-06-10:
Half of mine is Digikey, the other half is Mouser, and some odd parts were eBay purchases. I noticed that the original BOM is almost all either deprecated, obsolete, or straight up unobtainium though various parts dealers. I will throw a quick and dirty list together first thing tomorrow.

jeff464, 2024-06-10:
@circuitcreature I attached my BOM as of today. Please note that I have yet to verify it as working in a build, so take it with a gain of salt. I do plan on swapping the resistor network with the higher rated one from the other BOMs. Also, there might be other parts that don't make sense. If anyone notices anything weird, just let me know and I'll make a change. 

[NESizer BOM 06.10.2024 (NOT VERIFIED).ods](https://github.com/user-attachments/files/15774753/NESizer.BOM.06.10.2024.NOT.VERIFIED.ods)

jeff464, 2024-06-18:
[image]

So I think I may have screwed this up - I finished my NESizer build, but the headers don't seem to be making a proper connection. I don't get any activity when I power on, but if I squeeze the board down, I get *some* LED flashing. Looking at the updated BOM, I believe I ordered the correct part (649-68602-116HLF). Can someone verify that I totally biffed this?

I think the best option here might be to order longer headers and cut them down. Thoughts?

circuitcreature, 2024-06-18:
@jeff464  check your solder on those pins. I'm guessing it could be that, if that's all good the pins could be too short. You could try offsetting one side to try and center the pins between each connector.

beau-seidon, 2024-06-18:
Yes try to get the male pins centered between the boards so they make a snug connection in both sets of sockets

If that doesn't work: for testing purposes, detach your boards from the standoffs and connect them with breadboard jumper wires if you have any. Just for the time it takes to load the software and test functionality, to rule out other issues.

If there's extra room between the boards, you could try to carefully trim or sand the standoffs to a shorter height (assuming they are entirely plastic without metal thread inserts), but you'll have to do it carefully so they are all the same, and not too short such that the voltage regulator touches the top panel.

jeff464, 2024-06-18:
Solder tested okay with the multimeter, so I think I'm okay there. I thought maybe I installed this wrong since others seem to have the same size header without connection issues. That said, they really do seem a little too short, so if the components seem to be installed okay in the photo, I'll just swap the headers with something much longer. Good call on sanding the standoffs and using jumper wire, though. Unfortunately I don't have more than a few wires, and the center of the standoffs are metal. 

I attached another photo showing the length discrepancy. This is odd if others got it to work.
[image]

jeff464, 2024-06-18:
I just realized I have snipped leads from the LEDs I could probably use temporarily. ;)

beau-seidon, 2024-06-18:
I just checked my order history, and yes [649-68602-116HLF](https://www.mouser.com/ProductDetail/Amphenol-FCI/68602-116HLF?qs=PCEjGSkOlsa%252B8AOHtx6MSg%3D%3D) is the part number I used for the 2x8 pin connector.

What is the part number for your standoffs? I wonder if they are ever so slightly longer than mine.
I opted for the plain nylon 25 mm [144-CBTS115A](https://www.mouser.com/ProductDetail/Essentra/CBTS115A?qs=PzGy0jfpSMt5%2F9rQWt9IYQ%3D%3D).

[200-TSW10808TD](https://www.mouser.com/ProductDetail/Samtec/TSW-108-08-T-D?qs=rU5fayqh%252BE0hIpjMtL674Q%3D%3D) may be a better male connector for this, the termination posts are longer (5.08 mm vs 3.81 mm).

beau-seidon, 2024-06-18:
I just checked my build photos and noticed yours is put together differently from mine. You soldered both female connectors to the boards, whereas I soldered the male header to the bottom board and used the female+male with the long legs as the middle piece.

[image]

beau-seidon, 2024-06-18:
Smash it and start over. The one on my BOM is one of 2 parts not available lol. 

The part I used was 992-8FX1L-254MM, but that one doesn't even have a page anymore on Mouser. Probably something to do with politics.
I can't even find a datasheet for it, the manufacturer website is gone and everything. 

I managed to get some info via wayback machine and this part should be equivalent: [801-87-008-53-001101](https://www.mouser.com/ProductDetail/Preci-dip/801-87-008-53-001101?qs=q5BRkYolQ3gnRfYCsKSECg%3D%3D)
Or you could just search "stackable header" and choose your favorite. The male pins are ~12 mm.

jeff464, 2024-06-18:
Yep. Just threw the build in the trash. Oh well!

Standoffs look to be 25mm, but the photo confirms what I feared - I placed the components on the wrong way. The weird part is that I never had to snip the leads on the female sockets, so I would imagine it's all the same length anyway if all three parts are the same, BUT I guess not lol. Not a huge deal.

Thanks for the quick research. I sincerely appreciate it and the info helps speed things up a bit. Super weird that the original part vanished from Mouser. What would even cause that? 0_o

beau-seidon, 2024-06-18:
Dunno. There should have been 2 of these long-legged gals:
[image]

But since they're already glued in place, rather than messing around chopping parts off and resoldering, just get another longer male header like this to connect them:

> [200-TSW10808TD](https://www.mouser.com/ProductDetail/Samtec/TSW-108-08-T-D?qs=rU5fayqh%252BE0hIpjMtL674Q%3D%3D) may be a better male connector for this, the termination posts are longer (5.08 mm vs 3.81 mm).

jeff464, 2024-06-18:
Awesome! Thank you. Side note - I went out and bought some breadboard jumper wires and connected the PCBs. The unit turned on with solid LEDs for a few seconds which I assume is some sort of initialization process, but then the LEDs/display started looking like they are twitching. Nothing is working, but I'll have to test again later once I buy a coin battery. I'm thinking that's the main issue now.

jeff464, 2024-06-19:
Okay, so I bought a battery and the unit is still misbehaving. I will now move this over to the "Build Problems" thread.