Jaffe-/NESizer2 #11: AVR Programming Method

beau-seidon, 2023-05-26:
I'm new to interacting on GitHub, so my I apologize if it's poor etiquette to create new issues on your project for basic questions like these. 

My intention is to stick the Atmega328 on a breadboard with the bare-minimum required components as shown in the Breadboard Arduino reference here:
https://docs.arduino.cc/built-in-examples/arduino-isp/ArduinoToBreadboard
and then power and program it via my Raspberry Pi's GPIO / SPI pins using avrdude before installing. 

Is that what you'd recommend or is there a better way?

beau-seidon, 2023-05-26:
I also have an Arduino Uno board I can drop the Atmega into, which would be much simpler and my preferred method if it would work,

Jaffe-, 2023-05-27:
Hi Beau,

The way I have been doing the programming is to just have the Atmega328 seated in the board and use the 6 pin header (SV1)  on the board to program it. The pin header provides an ISP (In-circuit Serial Programming) interface directly to the Atmega. I've used an old ICE programmer for that, but it should be pretty straight forward to use a Raspberry Pi as an ISP programmer: https://learn.adafruit.com/program-an-avr-or-arduino-using-raspberry-pi-gpio-pins/installation

There are probably many ways to go about this, but the reason why I recommend to just keep the Atmega in the board when programming, is that the Atmega will be programmed with new fuse settings which change the clocking from internal oscillator to using an external clock. So after the first time the Atmega has been programmed and gotten new fuse settings, it will not be possible to program again without an external clock running. Since the NESizer already has that clock there, it is easiest to just keep the Atmega in the board.

beau-seidon, 2023-05-27:
Perfect. That's even simpler and makes it less likely for me to screw it all up. 
The external clock I was going to use is a 16 MHZ crystal with caps, whereas yours uses a 20 connected to an inverter.
I was concerned about the clock difference which is why I asked.

Also that's one of the links I have bookmarked for programming reference, so at least I'm on the right path there.

Thank you!

beau-seidon, 2023-05-27:
Would it be a good idea to assemble everything and then program the Atmega before inserting the 2A03 in its socket, or does it need to be in there when I program it?

My biggest priority is to protect the 2A03, since it is the only precious component.

Jaffe-, 2023-05-27:
You can program it without the 2A03 in there. I think all you would need on the board to just program the Atmega is the power supply, ISP connector and clock circuit.

Note that you can not test that the Atmega is actually working correctly until a 2A03 is inserted, because the initialization code is going to hang if it does not get contact with the 2A03. But if you manage to program it with avrdude and set the fuses, than you can be very sure that the Atmega itself is going to work.

Jaffe-, 2023-05-27:
I think taking it out and using a 16 MHz crystal would not work, because the fuse setting is to use an external clock on XTAL1 - not external crystal with capacitors connected across XTAL1 - XTAL2. 

The inverters in the circuit diagram are there to make a clock signal together with the crystal and capacitors. It is a typical and easy way to make a clock generator circuit. The reason why this is done and not connecting the crystal and capacitors directly to the Atmega, is that the same clock is also used to clock the 2A03, so that the two can be exactly synchronized.

If you made a full clock generator circuit around the 16 MHz crystal and connected it to XTAL1, then I think it would also work fine to program it on an external breadboard.

beau-seidon, 2023-06-22:
When I set the ATmega fuse bits, I used the following settings:

    -U lfuse:w:0xe0:m -U hfuse:w:0xd9:m -U efuse:w:0xff:m 

which, for ATmega328P, corresponds to:

    Ext. Clock, Start-up time PWRDWN/RESET: 6 CK/14 CK + 65 ms; [CKSEL=0000 SUT=10] and no internal clock division.

Other than the the External Clock, and no division, I left everything else default. Does that sound right to you? 
Obviously, it's at least close, because the NESizer is working, but I want it to be right.

I used the following calculator to determine what arguments to pass to avrdude
https://www.engbedded.com/fusecalc/

Jaffe-, 2023-06-22:
Yes, that is the same settings I have used!