Jaffe-/NESizer2 #13: Build Errors

circuitcreature, 2023-08-27:
I have the synth all built and as mentioned in another issue I used the MXD1210CPA+ instead of the DS1210. I am able to flash the Atmega using an Arduino and set the fuses to  E:FF, H:DF, L:E0, or  E:FF, H:D9, L:E0.

The issue is when I power it up all the LEDs light up and it doesn't produce any sounds. I have also tried the init within the settings using buttons 8,9,11, and 14, but nothing appears to happen after pressing 11 and 14. Pressing 8 then 9 does cause the LEDs to brighten and a sequence starts on the screen for about a second.

I can however test the chip type correctly with a returned value of 12. 

Once after powering on it did start making sounds, but I haven't been able to reproduce that.

Any troubleshooting ideas?

Jaffe-, 2023-08-27:
It is normal for it to behave strangely at first bootup, because the RAM just contains junk. The RAM is initialized properly when going to the settings mode and using 8, 9 and 11. Next boot after this it should work more as expected. 
This is one of the less finished parts of the SW, it should really just auto detect corrupt memory and initialize automatically.

If it it does not help to press, 8, 9 and 11 (when in settings mode) and power cycling, then I suspect that something might be wrong with the SRAM memories or surrounding circuitry. I would check the RAM chips (IC10, IC11) and the address latches (IC7, IC8, IC9) and the address decoder IC5 for any soldering issues. Same for the battery backup circuitry (DS1210). And of course, just to double check, you do have a battery installed?

circuitcreature, 2023-08-27:
I put the battery in and tried the init and power cycling and everything is working. I still have the issue with the LEDs always being lit, but I can live with that.  I'll have to check my BOM, but I ended up using a different LED from the committed BOM, so there might be something in there. 
Great project thank you.

Jaffe-, 2023-08-27:
That sounds strange. Are they always on, but brighter when actually supposed to be lit? Otherwise it sounds like it would be hard to use ...

I pushed some new code now, it checks whether the RAM has become corrupted. If it is, 'cr' will flash in the display at boot and the RAM will be properly initialized (same as manually performing it in the settings menu).

circuitcreature, 2023-08-27:
Correct, they are alwasy on and sometimes get brighter, this is only noticeable when the init runs after pressing 8 => 9, but other than that there is no indication when on a specific mode. I used mouser #696-SSL-LX3054IID, but they look to be mostly the same as the OG BOM.

Thank you for the code update. I'll give that a try next time I get the flasher setup.

Jaffe-, 2023-08-27:
Sounds like something might be wrong on the panel board. I would check the row and column latches there. Do all the LEDs do this, or just some of them? What about the 7 segment display?
The LEDs are very similar to the ones in the BOM, so I don't think they are the issue here.

circuitcreature, 2023-08-28:
Double checked everything and it looks fine. It only happens to the LEDs, the display displays 00 when nothing has been selected. 
As for the update, I get a battery warning and cl on every boot and it wipes all settings.

beau-seidon, 2023-08-28:
There are a few items in the committed BOM which are obsolete. I had to select suitable replacements. Also @Jaffe- did mention, in a reply to one of my Issues, that the BOM is not as up to date as the schematic. One example I found is that the part number listed in the BOM for the bus resistor on the top board is for a 4.9k instead of the 15k from the schematic. 

When I built mine, I opened the .sch file in Eagle (free version) and exported a new parts list, then carefully compared the committed BOM to the new list. I used both as a reference to build my own cart on Mouser, defaulting to the schematic when I found conflicts. 

I have attached an export of my Mouser cart with all the parts I used in my build, for you reference. 
Mine is working great without any electronics issues.

[exported_mouser_cart_(nesizer_only).xls](https://github.com/Jaffe-/NESizer2/files/12450320/exported_mouser_cart_.nesizer_only.xls)

circuitcreature, 2023-08-28:
Thank you @beau-seidon I'll compare yours to mine and update as needed. I'll report back if I order any new parts.

Jaffe-, 2023-08-28:
> Double checked everything and it looks fine. It only happens to the LEDs, the display displays 00 when nothing has been selected. As for the update, I get a battery warning and cr on every boot and it wipes all previous settings, like midi channels.

Does not sound good! I think the cr error should only occur once the first time (and yes, the settings will be erased too. I should have warned about this...) After it has written the expected signature to RAM, it should work fine after the next boot. If it does not, it means that the battery backup is not working and has corrupted the RAM.

I will check whether the battery fault is now always showing with a fresh battery also. It might need some more work to be reliable.

beau-seidon, 2023-08-28:
> Thank you @beau-seidon I'll compare yours to mine and update as needed. I'll report back if I order any new parts.
> edit:
> Ordered more parts, the resistor array, clock, and diodes. 

If you are talking about the switching diodes, you can use the ones in the BOM (1N4148 are equivalent to the 1N914 diodes that I bought, and there will be no functional difference).

The clock will make an enormous difference if you got the wrong one. Particularly if it oscillates at a different frequency than the design. If the freq. is wrong though I wouldn't expect it to boot at all. 

Not sure what effect the resistor array difference would make, other than potentially passing too much current through the buttons if several are pressed at once? Full disclosure: I bought the wrong one too and almost said screw it I'll install it anyway lol. In the end I decided not to and got a 15k replacement before turning it on.

I'm not implying anything here, but triple check that all your IC's are oriented correctly. I booted it up the first time with my 2A03 in backwards because I am stupid, but it didn't hurt anything as far as I can tell. It still goes beep boop.

circuitcreature, 2023-08-28:
ICs look good. I also picked up more latches, to test out. I might have messed something up on first boot, i had the pins off by one with the mainboard and panel. As far for the cystal it is 20mhz, worse case I'll build another one.

Jaffe-, 2023-08-30:
@circuitcreature  Let me know what you find out. If you know some programming, you could try to change the update speed of the LEDs in the code. See src/task/task.c line 84 which defines the update rate for the LED refresh. If you increase this period, it should update slower and perhaps make it easier to debug. You could also try and just locally mod the code to add a mechanism to single-step the LED rows and columns from one of the buttons.

Regarding the battery level and corrupt RAM warnings, it just dawned on me that the board I currently have fully assembled is an older v0.5 board with some mods, so the battery related stuff is not the same as in the latest design. I should build a new main board based off the latest version before attempting to write any clever code, I think. I will revert the change related to battery low and memory corruption errors until I can test it on a new board.

Just out of curiosity, what do you see when you press the battery level indicator in settings? (press and hold second rightmost button on first row)
Same for you as well, @beau-seidon, what does the battery level indicator show in your case?

circuitcreature, 2023-08-30:
@Jaffe- I've been digging around in the code, it's been a while with C. I'll look at the refresh rate.
As for the battery it displays 45. I'm pretty sure this one is new, I can dig around and see if I have another to test.

beau-seidon, 2023-09-02:
> Just out of curiosity, what do you see when you press the battery level indicator in settings? (press and hold second rightmost button on first row) Same for you as well, @beau-seidon, what does the battery level indicator show in your case?

I haven't updated my NESizer with the most recent code you pushed. 
It is running a hex with my NoteOn&Vel=0 changes, and a hack to mute the output at boot, but no other mods.

When I check battery voltage by selecting Settings mode, and holding Button 7, I get **4.3** on my segment LED.
I can update the firmware with your most recent code and check again if you'd like.

beau-seidon, 2023-09-02:
> Regarding the battery level and corrupt RAM warnings, it just dawned on me that the board I currently have fully assembled is an older v0.5 board with some mods, so the battery related stuff is not [...]

@Jaffe- I have 4 more pairs of the latest PCBs in red+blue. You want one?

circuitcreature, 2023-09-06:
I changed the speed on the LEDS and all they do is pulse slower, but still remain on.
I've swapped out the latches on the panel board and no change with the LEDs. I also swapped the latch and the ICs next to the atmega chip and no change. Out of curiosity I removed an LED and swapped its polarity and it remained on.  

[image]

beau-seidon, 2023-09-06:
How did you orient your LEDs? 

When I built mine it wasn't exactly clear without following the traces on the PCB, which was doable, but I found that the cathode/anvil for all LEDs should face the right side of the board. 
(In the photo below they point left because the board is rotated.)

@Jaffe-  even if not the issue here, perhaps the silkscreen on the top board should be updated to show LED orientation in future versions to help prevent mistakes by new builders.

[image]

Jaffe-, 2023-09-06:
That is right, the cathode is the flat part of the epoxy. Also before cutting the pins, the short one is the cathode on the LED. 

The silkscreen is unfortunate as @beau-seidon pointed out. It is actually indicated on the silkscreen layer if you look at panelboard_pcb.png, on the right side the circle has a flat/straight edge. However, on the produced board (at least my ones) this is not possible to see because the holes cut through it. I agree this should be more clearly indicated!

The correct orientation is that the cathode should be on the right side when viewing the board the same way as in panelboard_pcb.png.

circuitcreature, 2023-10-07:
I had some time to work on this and got it working. The original LEDs I had were the issue. After a long slow boat shipment, I replaced them with another brand and everything works great. 

I also tested with a 2A03-G and a UA6527P, and both work great.

I've seen the recent commits and will try that version soon.

Thanks @Jaffe- and @beau-seidon !

[image]

beau-seidon, 2023-10-07:
Nice! I'm glad you got it working. 

It's tied with the Twisted Electrons TherapSID for my favorite chip synth.

circuitcreature, 2024-03-22:
@Jaffe-  Not sure if you can help. I went to turn on the synth the other night and there was no sound or lights. I've checked the voltage across all the chips on the panel and several on the main board and most everything is showing 5v. 

The odd thing is that avrdude programs the chip just fine.

Any advice on things to check?

Jaffe-, 2024-03-25:
@circuitcreature: That is weird! Have you checked the clock and its power supply too?
This might also be something that is failing during initialization. Everything will hang if the 2A03 is not working for instance. What kind of 2A03 do you have installed?

circuitcreature, 2024-03-26:
Pin 1 on the clock is 2v and pin 2 is 2.5v
I have tried both a RP2A03-G and UA6527P

Jaffe-, 2024-04-09:
@circuitcreature That is strange indeed. I'll try and see if I can decouple the 2A03 code a bit from the system so it can boot without it installed (probably a nice feature anyway, at least the NESizer can then be used as a MIDI sequencer :p)

In the mean time, can you try switching out the Atmega and see if that makes any difference? I did have an issue with an Atmega earlier that was somewhat like what you describe, but I thought it was just a bad chip because it worked fine when I switched it out...

Jaffe-, 2024-04-09:
@circuitcreature I added support for running without 2A03 now. You can try building and installing that commit to see if the communication with 2A03 is the issue (although it seems unlikely as it fails with two different chips).

beau-seidon, 2024-05-04:
try re-burning the fuse bits. I temporarily bricked my atmega chip by accidentally pressing ctrl+c in the terminal while it was flashing firmware. couldn't get it to do anything or respond to another flash command, but found that re-burning the fuse bits reset it and it's working fine again.

circuitcreature, 2024-05-06:
Thanks @beau-seidon. I've swapped chips and finally got something, even though I think I'm back to the original chip that first started with the issues. I have it setup for debugging and it's powering on, but not functional. It could be that it's connected with test wires. I'll try again when I have some more time. I also have some more atmegas in the mail, I'll be able to test with those if all fails.

beau-seidon, 2024-05-10:
I just got a UA6527P from eBay to test it out and it doesn't seem to work at all in the NESizer. 
My 2A03G from my NES still works perfectly. Might just be a bad chip. No idea how to troubleshoot it.

beau-seidon, 2024-05-15:
It did not work in my NES. I contacted the seller on eBay and he refunded me. 
I'd rather have a working chip than the money though, that's why a bought it lol.

I'll test it for fun using your function when I get home, just to see what it says.

circuitcreature, 2024-05-15:
The 6526p does work. I have a couple more i can test when I'm back up and running. I'll have some atmega chips soon. 
@beau-seidon  if this batch of 6527p works i can send you a link.

beau-seidon, 2024-05-20:
> What do you see in the settings page if you press the 'CHIP TYPE' button (rightmost in the upper row)? If it says 0, then I would say that it is a high probability that the chip is not functioning correctly, because this means that it was not able to complete the 2A03 detection sequence (feeding it instructions and checking for R/W line transitions). 

Using the latest build, pressing the CHIP_TYPE button causes the 7 segment display to show [ - - ], when the button is held.
With my 2A03 chip in, it shows [ 1 2 ], which I expected since I removed it from my NES. 
So yeah, the UMC chip is either dead, or fake.

beau-seidon, 2024-06-19:
This is a reply to: https://github.com/Jaffe-/NESizer2/issues/10#issuecomment-2177410174

@jeff464 please describe how your unit is "misbehaving". No lights or flickering lights? No response to button input?
Were you able to successfully set the fuse bits, and flash the firmware to the ATmega chip?  

Which software are you loading to the NESizer? 
I recommend using the @Jaffe- master branch, until you are sure it's working, then play with experimental forks/branches. 

If you have trouble compiling or uploading, try using VSCode+PlatformIO to build/load it. If you use the prebuilt .hex file, you can use avrdude (comes with ArduinoIDE) to flash it.  For more info see this link, and check out the other Closed issues: https://github.com/Jaffe-/NESizer2/issues/24#issue-2302456124

Funny noises coming from output channels?
I know the software has been updated a few times since I built mine, but this post might still be relevant, regarding first-time startup: https://github.com/Jaffe-/NESizer2/issues/7#issuecomment-926837146

Lights and input ok, but no sound at all? 
Make sure you have the voices you want to play enabled, and assigned to MIDI channels. 
Also make sure your audio cable is in the right jack and the switch is set properly if you are mixing to mono.

jeff464, 2024-06-20:
I turn on the unit and some lights flicker/character display barely flickers.
Unit is unresponsive to any button pushes/no sound. 
All points checked out okay with the multimeter, 
Official .hex was used and wrote successfully/verified using GQ4x4 writer.

Other notes:

Command line anything isn't an option for a few more weeks - trying to maintain my sanity (build/troubleshooting now going on 90+ hours, timed)
Not entirely sure how to program fuse bits using the GQ4x4. I assume this is the root of the problem.

jeff464, 2024-06-20:
Tried using avrdude since I'm a glutton for punishment - it opened and immediately closed despite running as administrator. Not ideal, so I found a GUI by the name of avrdudess. Cool. 

I now have an arduino uno, atmega328p, and avrdudess. However, I cannot for the life of me find instructions on how to upload the hex + fuse settings to the 328p using just the uno. Surely this is possible?

jeff464, 2024-06-20:
Thank you for the detailed response - I sincerely appreciate it. I ended up flashing the Uno board, went out and bought another, came back, wired the two Arduino Uno boards together, and used AVRDUDESS to write the .hex w/ fuse/lock bits. Unfortunately, I am receiving the same exact behavior as before. I'm going to attempt to attack this from a few different angles over the next few hours.

beau-seidon, 2024-06-20:
Are you programming the NESizer ATmega328p chip by installing it in an UNO/other external board?

Try putting it back in the NESizer and connecting the UNO configured as an ISP programmer directly to the header on the left side of the NESizer. 

If you are programming it while loaded in an UNO board with the 16 MHz crystal, I'm not sure how it will behave if the fuse bits are set such that the chip expects a 20 MHz clock. This was my plan originally but I never tried it, I just programmed it "In-System".

jeff464, 2024-06-20:
I connected the side ISP pins directly to the Uno and force flashed - avrdudess wasn't recognizing the 328. I'm getting a similar result with the LEDs/display, but at least now it's responsive to a degree. When I hit the settings button, the LED lights, so that's a good sign.

That said, I'm reading about the NESizer initialization button combination in order to clear the garbage data, but I'm confused. It says to press button 8, 9, 11, and 14 while in settings mode. Is this at the same time or in order? If in order, how long do I wait between button presses? Also, those button numbers don't align with the numbers on the PCB, so I'm not sure if it actually means 9, 10, 11, and 15.

jeff464, 2024-06-20:
This is what the display looks like when powered on.
[image]

beau-seidon, 2024-06-20:
It's alive! (More alive than it was anyway :) )
Press them one at a time, in the order @Jaffe- stated in that [post](https://github.com/Jaffe-/NESizer2/issues/7#issuecomment-926837146).  
The PCB silkscreen is a little out of date, and apparently so is the Settings section of the manual. This is from ui/ui_settings.c
[image]
[image]

The first few go quick, but the last one takes forever because it formats the big memory chips. Give it about 5 minutes of SAMPLE_FORMAT then reboot it, and it should work!

jeff464, 2024-06-20:
Hmmmm, I think something is up. I go to settings, hit 8, and then the four of the bottom LEDs flash rapidly. I assume this is supposed to happen, but nothing changes after about ten minutes lol. 9 seems to do something - the character display rotates briefly. 11 does nothing, and 14 also seemingly does nothing after several minutes.

If I plug in the audio cables and crank the volume, I can hear a quiet whine and tapping.

beau-seidon, 2024-06-20:
If you reboot does it work now? Does the 7seg display 00 and change with the Up and Down buttons?
If so push one of the voice buttons (0 - 5). Does the corresponding LED stay lit? 
If so, go to Settings mode then hold MIDI Channel then select one of the voices and choose a channel, then feed it notes.

jeff464, 2024-06-20:
It does not work after resetting, and the display does not change. Pressing channel 1 lights LED 1, 3, and 8. Pressing channel 2 lights LED 1, 2, 3, and 8. Pressing channel 3 lights LED 1,3, and 8. Pressing channel 4 lights LED 1, 2, 3, and 8. Pressing channel 5 lights LED 1,3,5, and 8. They do stay on once pressed. If I choose a a MIDI channel, LEDs flash, but I have to reset the unit for it to stop flashing. There's no audio, and I still hear clicking/whining.

beau-seidon, 2024-06-20:
I see you have the top and bottom boards connected via jumper wires. I wonder if there's a bad connection there. 
Also when you flashed the .hex using your UNO, did you first set the fuses with the chip in the NESizer?
( Low = E0, High = D9, Extended = FF )

jeff464, 2024-06-20:
I now cannot get any software to read (or write without force) the 328, so I'm going to assume it's bricked/corrupt.

beau-seidon, 2024-06-20:
> After your UNO is configured and your connections are made, launch Windows command prompt and paste this command in to burn the fuse bits (replace COM4 with the COM port for your own UNO):
> 
> ```
> avrdude -C "C:\Program Files (x86)\Arduino\hardware\tools\avr\etc\avrdude.conf" -v -p atmega328p -c stk500v1 -P COM4 -b 19200 -U lfuse:w:0xe0:m -U hfuse:w:0xd9:m -U efuse:w:0xff:m
> ```
> 
> It will launch avrdude, tell the application where to find the config file, tell it to print out what it's doing, tell it what kind of chip it will be programming, what programmer it will be using, and which COM port, baud rate, and the fuse bit values to write.
> 
> You may have to re-load the .hex file to the board afterward. If so, you already know how to do that. If you want to upload it with avrdude using your UNO use this command (update the COM port and path to your .hex file before executing):
> 
> ```
> avrdude -C "C:\Program Files (x86)\Arduino\hardware\tools\avr\etc\avrdude.conf" -v -p atmega328p -c stk500v1 -P COM4 -b 1
> ```

Try this ^. Use the command line. Copy and paste those commands to a text editor, update your COM port and .hex path, connect your programmer, then paste/run those in CMD.

I oopsed my chip once by accidentally disconnecting while flashing and it wouldn't do anything anymore. Couldn't flash a new program either. I fixed it by re-burning fuse bits, and then it would let me reprogram again.

jeff464, 2024-06-20:
I'm just getting this as a return:
`avrdude: Version 5.10, compiled on Jan 19 2010 at 10:45:23
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr\etc\avrdude.conf"
syntax error at C:\Program Files (x86)\Arduino\hardware\tools\avr\etc\avrdude.conf:356
`

beau-seidon, 2024-06-20:
does that path exist and is there an avrdude.conf file in there?

I just noticed the ArduinoIDE on my work pc is not installed in "C:\Program Files (x86)".  On my laptop the path above is correct.
I found the file at the path `"%LOCALAPPDATA%\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17\etc"` by compiling and uploading a sketch in ArduinoIDE with verbose mode on. 

You may have to update your path to the .conf file.

jeff464, 2024-06-20:
Heh. Yeah, PlatformIO it is. The path is correct, but it's now turning into a side quest like I feared. Downloading now.

jeff464, 2024-06-20:
Ok, I opened the nesizer.hex and this is what I'm seeing. This doesn't seem right at all.
[image]

jeff464, 2024-06-20:
Maybe it's time I just give up on this build forever lol

beau-seidon, 2024-06-21:
> Maybe it's time I just give up on this build forever lol

Nah, finish the build and get it working! Then give up.

> Ok, I opened the nesizer.hex and this is what I'm seeing. This doesn't seem right at all.

Yeah, that is what .hex files look like inside. It's compiled machine code. The ATmega knows what to do with it. It's not for you.

Go to the home page of this repo, then click the green `<>Code` button, and click [Download Zip](https://github.com/Jaffe-/NESizer2/archive/refs/heads/master.zip). Extract the contents and right-click on the sub-folder called NESizer2-master (with docs, hardware, src, etc. inside), then in the right-click menu choose `Open with Code` (meaning VSCode).  The PIO extension should recognize the platformio.ini file inside, and automatically download all the required AVR backend stuff. 

Open that .ini file from the Explorer panel on the left and change `upload_port = COM5` to your own COM port, and save it. 
Connect your programmer to the NEZizer, then click on the PIO icon (alien head on the left side bar). There should be a Project Tasks menu that appears at the left. Expand and click `Project Tasks / ATmega328P / Platform / Set Fuses`. This should burn the fuse bits with the configuration in the .ini file, and get the chip working again.

Next build the project using the check mark button in the top right, and you will see a "parameter name omitted" compile error :'( (I'm about to put in a pull request to fix this). 
To fix the error yourself, open `src / io / 2a03.c` and replace line 186 with `void register_set_noop(uint8_t a, uint8_t b)`. 
Then build again. Should be error-free now! (You can skip the first build step and just fix the bug first if you want...)

Next to the Build button there is a small drop down arrow, click that and select Upload. It should load the compiled firmware to your board while printing a bunch of progress messages (hopefully instead of errors) in the console.

Let me know if that helps!

jeff464, 2024-06-21:
Thanks again for the assistance. I feel like I should be paying you. Usually I'm not this awful at DIY, I swear.

I'll follow these instructions first thing tomorrow morning. Looking forward to building this thing and then giving up. ;)

jeff464, 2024-06-21:
First snag - there was no "open with" in the right-click menu for that folder. Obviously not a huge deal. I just opened the folder in VSC and everything seemed to populate. Cool.

Second snag - set fuses returns an error:

Selected fuses: [lfuse = 0xE0, hfuse = 0xD9, efuse = 0XFF]
Setting fuses
avrdude error: syntax error [C:\Users\Unknown:1]
avrdude error: unable to process system wide configuration file C:\Users\Unknown
*** [fuses] Error 1
=============================================================================== [FAILED] Took 3.92 seconds ===============================================================================
 *  The terminal process "C:\Users\Unknown User\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'fuses', '--environment', 'ATmega328P'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
 
 Something *REALLY* doesn't want me to build this thing.

jeff464, 2024-06-21:
Thinking about this, it might have to do with the user folder having a space in the name. "Unknown" is only part of the username/folder, and the error seems to throw at that space. Short of editing the win registry or creating a new admin account/porting over all installed apps and programs (not a chance) I think this might officially be outside of the scope of NESizer. What a nightmare.

circuitcreature, 2024-06-21:
It's been awhile since I've worked in the Windows CLI. You might want to try putting the paths in double quotes. e.g. "C:\User\Unknown ..." 
You can also use the tab key to auto complete paths when typing and that should auto add the quotes.

beau-seidon, 2024-06-21:
> Thinking about this, it might have to do with the user folder having a space in the name. "Unknown" is only part of the username/folder, and the error seems to throw at that space

Yep looks like you found a bug in the python scripts that do the fuse burning and uploading. It seems to be adding extra quotes around your path which cancel out the original quotes that allow spaces in the path string. According to the PR for the issue linked below, the scripts were fixed, and I checked those installed on my system and mine reflect the PR changes. 

https://github.com/platformio/platform-atmelavr/issues/297

Apparently, it's not entirely fixed. Since it is still buggy on your system, go find that config file in 

`C:\Users\"your user name"\.platformio\packages\tool-avrdude\avrdude.conf`

and copy it to a temporary folder in `C:\` (root directory).

Then replace line 30 in platformio.ini:
`    $PROJECT_PACKAGES_DIR/tool-avrdude/avrdude.conf`
with your path to that file. Something like:
`    "C:\tmp\avrdude.conf"`

jeff464, 2024-06-21:
Now I get a different error. This is the same error I was getting before when I realized it was no longer working. Everything is definitely hooked up properly.

Selected fuses: [lfuse = 0xE0, hfuse = 0xD9, efuse = 0XFF]
Setting fuses
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude error: Yikes!  Invalid device signature.
avrdude error: expected signature for ATmega328P is 1E 95 0F
        Double check connections and try again, or use -F to override
        this check.

avrdude done.  Thank you.

*** [fuses] Error 1

jeff464, 2024-06-21:
I found a post that says the error is coming from shared ISP pins, which *might* check out here considering I'm getting the same error as before when writing to the 328 when it's in-circuit with the NESizer.

https://forum.arduino.cc/t/burning-bootloader-expected-signature-for-atmega328p-is-1e-95-0f/88295/16

But really, who knows at this point,

beau-seidon, 2024-06-21:
For the sake of thoroughness, would you mind posting a pic clearly showing the letters on top of the ATmega328 chip?

jeff464, 2024-06-21:
Of course!
[image]

[48 more replies]