Jaffe-/NESizer2 #15: Midi CC integration

circuitcreature, 2023-11-17:
I already have this implemented on a branch and need to run a few more tests on it before I make a PR. Just creating an issue so that it can be tracked for any bugs.

beau-seidon, 2023-11-18:
This is very exciting. I was actually going to start work on implementing midi cc this week, but you beat me to it. I was going to try to stick closely to the implementation of the Famimimidi, but I do understand that the two devices work differently. 

I installed your updated firmware last night to try it, and had some fun fiddling with cc params while the sequencer was playing.

I did find that with no midi channels assigned to the voices (all default of 0), that my Nesizer freaked out and would lock up and become unresponsive until a reboot. I was able to reproduce the problem, and the solution was the assignment of midi channels to the voices.

circuitcreature, 2023-11-19:
Ha, I was going to match the MidiNes, but ended up trying to match the midi cc defaults. I was thinking about creating a map with some different options to match other like Famimimidi and MidiNes. 

I was able to crash mine as well, but wasn't sure if it was my controller. I'll get an update pushed soon.

jeff464, 2024-06-10:
Looking at this thread - is CC integration now available on the main NESizer build, or is it only just available in the forked build as of now? I'm curious because I have a Famimimidi and this functionality would go very well with my current MIDI templates. 

Also, since you're familiar with the Famimimidi  - are you able to decipher any of the MIDI commands/instructions regarding macros/patch storage? I was never able to figure this out. Way over my head!

circuitcreature, 2024-06-10:
@jeff464 CC integration is only on the forked branch right now. My last commit is untested but should work fine. Im only familiar with Famimimidi's documentation on it's midi protocol, but the more I dug into it, the more it matched the standard midi maps for CC, so I started to just map CC to those standards.
Actually if you test this branch and have any pointers to have it work more closely to the Famimimidi, feel free to commit updates. I'm hoping to have my Nesizer working in a few weeks and I can test and make any code improvements.

beau-seidon, 2024-06-10:
> Looking at this thread - is CC integration now available on the main NESizer build, or is it only just available in the forked build as of now? I'm curious because I have a Famimimidi and this [...]

I finally got my hands on a Famimimidi a few weeks ago, and have been playing with it a lot!

Aside: I actually found the NESizer project in my search for a Famimimidi alternative, because hekekekun quit making them so Catskull couldn't get them anymore and there have been none available on eBay or Reverb etc. for a long time. I still like the NESizer better, but there are some things it needs, such as short period noise (will post about this specifically in a discussion thread soon).

Some time in the near future I'll be testing the fork with CC implementation more and see how it compares fo Fmm

No idea how to do macro/patch storage on Fmm yet. I'll look into it.

jeff464, 2024-06-10:
@circuitcreature - I'd love to test this (actually, I'd prefer to). However, I spent over 48 hours last week trying to figure out why my Win machine doesn't properly compile the NESizer code as a .hex file. I never figured it out, and it's something I'll eventually troubleshoot next week or so. If you were able to upload a compiled .hex, it would speed things up a bit. I'll throw it on the 328 and test as soon as my unit is fully built this week.

jeff464, 2024-06-10:
@beau-seidon - Yeah, I feel the pain. I've been an early adopter of the MIDINES (back when the carts still had the white paper labels) and have been following all of the different NES MIDI carts pop up and immediately disappear over the years. I found myself in Japan back in 2019 and purchased a few different MIDI carts from hekekekun directly, but despite this, I've been meaning to build the NESizer for what seems like ten years now. I'm glad to FINALLY jump in. :)

As far as the macro/patch storage goes, I attached an English translated manual. Looking at this again, it seems the user macro stuff is at CC45-CC59. Also, there a section about sysex setting user wavetable data for the sample channel. I don't suppose anyone knows anything about that either? I tried to wrap my mind around it all a few times, but never really got anywhere. :(
[Famimimidi_implement_chart_v10_NES.pdf](https://github.com/user-attachments/files/15775153/Famimimidi_implement_chart_v10_NES.pdf)

beau-seidon, 2024-06-10:
> @circuitcreature - I'd love to test this (actually, I'd prefer to). However, I spent over 48 hours last week trying to figure out why my Win machine doesn't properly compile the NESizer code as a .hex file. I never figured it out, and it's something I'll eventually troubleshoot next week or so. If you were able to upload a compiled .hex, it would speed things up a bit. I'll throw it on the 328 and test as soon as my unit is fully built over the next day or two.

I had issues compiling using avr-gcc on Windows, but it worked perfectly on Linux. Have you tried WSL?
My recommendation is to use VSCode with the PlatformIO extension. That's how I do it now and it works great. Way easier than compiling via CLI.
Also @Jaffe- released the latest build in .hex format a few days ago, so you could also just download that and load it with avrdude.

jeff464, 2024-06-10:
@beau-seidon  Heh, yeah I reached out to Jaffe a few days ago when I was in the thick of troubleshooting avr-gcc. I requested a .hex compile, and he very kindly obliged. I apologize for any confusion - I personally don't have a fork of NESizer. In my previous message above, I was referring to a .hex compile of the NESizer version with MIDI CC enabled since (as far as I know) the official one doesn't have the CCs enabled.

I remember Jaffe mentioning the PlatformIO extension. That was next on my list to try. If that doesn't work, I'll likely openly beg for assistance here in the future. ;)

beau-seidon, 2024-06-10:
oops, I must have misread something, I'm at work, but will reread again later and fix my stupid.

circuitcreature, 2024-06-10:
@jeff464 I love my MidiNes. Grabbed one before Wayfar stopped responding to requests. This project spoke to me when i saw that you could upload samples to it.

I uploaded a release.[CC support](https://github.com/circuitcreature/NESizer2/releases/tag/Midi-CC)

I haven't setup PlatformIO to compile this version, but it should have compiled just fine.  I 100% recommend trying PIO out, it was the only way I could compile on my Win machine and it compiled perfect on my Mac even when the CLI was giving me grief.

jeff464, 2024-09-15:
Swinging back around to this - how are you all uploading hex files to the NESizer? Three hours in and it's still not obvious on how to do this with VSC/PlatformIO.

beau-seidon, 2024-09-15:
The way I found to load a precompiled hex file with PlatformIO is kind of hacky. It works, but it is way more straightforward to just use avrdude via the command line. If I recall correctly I saw a GUI wrapper for avrdude too, which may be easier to use.

I prefer to simply clone or download the latest version of the repository, then open it in VSCode with PIO and build / upload directly with the buttons in the top right of the editor.

If you go read my other posts about uploading via avrdude vs PIO and still can't figure it out, let me know (and give me a little bit more info about your setup) and I'll try to write you a script that will do it for you.

circuitcreature, 2024-09-15:
@jeff464 I can push up an update with PlatformIO setup with the midi cc. It's untested, but should compile and flash like the main branch

beau-seidon, 2024-09-15:
I submitted a pull request on your fork a while back to make it PIO compatible, It should function just like your main branch.
https://github.com/circuitcreature/NESizer2/pull/1

circuitcreature, 2024-09-15:
@beau-seidon Missed that PR. All merged and looks good. 
@jeff464 I have tested the Midi CC build using my PIO integration, but the updated master branch should would just the same.

jeff464, 2024-09-15:
I believe I successfully uploaded the MIDI CC firmware to the NESizer, but it seems MIDI CCs are not actually working on my end. Is there a trick to get this working?

jeff464, 2024-09-15:
Update - it turns out I didn't actually upload this firmware to the NESizer after all. PlatformIO iseems to be kicking back an error with the 2a03.c file
[image]

beau-seidon, 2024-09-15:
on line 186, change it to
`void register_set_noop(uint8_t a, uint8_t b)`

jeff464, 2024-09-15:
GREAT. That worked. Thank you! I'm now getting MIDI CC feedback from the NESizer, but it seems as though the CC adjustments aren't matching the original Famimimidi template I have or the CC descriptions in the original Famimimidi list. I will continue to test this throughout the day.

beau-seidon, 2024-09-15:
https://github.com/circuitcreature/NESizer2/blob/019c153959430aaf1363063cdc2b570633b6d0e1/src/midi/midi.c#L288

Check out the code in circuitcreature's midi.c starting at line 288 to see what CC numbers actually affect which NESizer parameters. It might not be exactly like the Famimimidi, which is of course ok, it is an altogether different toy. But the code itself will be a better reference than some other implementation sheet.

jeff464, 2024-09-15:
Got it. Thank you for the heads up. I was referencing this link and wondering why none of the CCs were working!

https://github.com/Jaffe-/NESizer2/pull/16

circuitcreature, 2024-09-15:
The PR references my first round of CC numbers, I caught myself last night when testing and spent too much time troubleshooting that before I remembered I needed to update my controller. If the CC update works for everyone i can type up a cheat sheet.

beau-seidon, 2024-09-16:
@jeff464 @circuitcreature 
https://github.com/circuitcreature/NESizer2/pull/2
I created another PR to incorporate recent changes/fixes that leaves the added MIDI CC functionality unaltered.
Best of both worlds!

circuitcreature, 2024-09-16:
All merged. I'll be able to test this out later this week.  I also been wanting to test out the sample play back

circuitcreature, 2024-09-17:
To keep things easy to configure the CC number are the same across each the channels.

I think for the effect you are shooting for, it would be better to keep the tracks on separate midi channels but triggering the same notes. You will have much more control of the sounds this way. 

Although it is interesting that the CC change isn't populating over the other channel when both are set to the same midi channel. 

If you want, you can create an issue request and that can be fixed/added in.

jeff464, 2024-09-17:
Yeah, that method was the alternative. There's a bit more faffing around with setup/creating separate templates per channel with the MPC, but if it works it works! One and done.

I'll create an issue request for the others. Thanks!

beau-seidon, 2024-09-25:
> Also, since you're familiar with the Famimimidi - are you able to decipher any of the MIDI commands/instructions regarding macros/patch storage? I was never able to figure this out. Way over my head!

After programming NESizer and my TinyMIDI2CV gizmo to handle SysEx over the past week, its now pretty clear to me how to load custom waveforms into Famimimidi and map them to PrgChg numbers. Do you have any particular shapes you want (saw, sine, something wild, etc)? If you send a screenshot of some waveforms, I can approximate them in 16x16 arrays, then I can send you SysEx files you can simply load to Famimimidi with MIDIOX or similar. Will explain how.

Also going to work on pitched waveform playback on NESizer soon. I think it's totally doable with an 8+8 fixed point phase-accumulator and wavetables. The task manager can probably handle it since it prioritizes the dmc channel. Can use the master ~16khz Tick as the sample rate. 

Just need to decide how to map it to MIDI. Maybe gonna make a 2nd loop mode in which you choose the waveform for each patch or upper/lower key group (using button UI or ProgramChange 100-127), and when you press the note, it decides how many elements to step thru the waveform array before starting over or how long to hold each element in the output register before stepping to the next based on the note you feed it. Made something similar on an Arduino ProMicro and it worked great.

jeff464, 2024-09-27:
Woah, that sounds WAY more in-depth than I initially thought. Please, do tell.  I don't have any particular waveforms in mind, but I would probably start with something like a supersaw lol. So I assume it's basically similar to how LSDJ handles custom wavetables, with the difference being that the user has to upload them to Famimimidi via sysex/midiox? Does the cart then store them?

This all sounds really exciting. I'll be updating to the latest NESizer firmware tonight. Stoked for those program changes!

beau-seidon, 2024-09-27:
Well the defining characteristic of a supersaw is that it is many normal saw waveforms playing simultaneously which are slightly out of tune so you get that wobbly chorus effect. 

You can't really get the same sound with a single monophonic wave period unless its a super long period. It's gonna just be a jagged 16 sample waveform, but I'll see how it sounds and send you the SysEx to upload (with an explanation of the contents).

beau-seidon, 2024-09-28:
[FamiWaveGen.zip](https://github.com/user-attachments/files/17172033/FamiWaveGen.zip)
This should help!

I created a little GUI with the tremendous help of ChadGDP, which will spit out a Famimimidi compatible SysEx string that you can paste into MIDI-OX. Draw the wave shape you want (like you would in LSDJ) and click Generate SysEx.

closed 2024-10-04 by beau-seidon

beau-seidon, 2024-10-04:
@circuitcreature 
The MIDI CC features from your fork were included in the recent PR. 

I originally intended to merge my own new features separately from yours because you had your own PR open for MIDI CC, but I had previously merged your CC into my master branch, and then added more stuff, and made the SysEx bug fixes etc., and it was a whole thing to factor out and I didn't see the point fussing around trying to separate everything, so when @Jaffe- said I could merge it, I just did it.

In future I will definitely ask/consult before merging and closing things I didn't create. 
I'll let you decide what to do with your MIDI CC PR, since your features are already incorporated but the PR now has conflicts.

circuitcreature, 2024-10-04:
@beau-seidon It's all good. What ever works easiest.