Jaffe-/NESizer2 #17: A recent update is causing a compile error:

circuitcreature, 2023-11-18:
There was a recent update in sample.c that is causing a compile error:

```
sample/sample.o: in function `.L22':
sample.c:(.text.allocate_block+0x176): relocation truncated to fit: R_AVR_7_PCREL against `.L9'
collect2: error: ld returned 1 exit status
```

[(https://github.com/Jaffe-/NESizer2/blob/47f1c84201cfba7b5e93a79f0b9ed0e8b224490f/src/sample/sample.c#L289)]

Jaffe-, 2023-11-18:
I don't get the same error on my end, which version of gcc are you using?

Jaffe-, 2023-11-18:
I wonder if this could be related to the change from -O3 to -O2 which was needed to fit in the new sample code and leave room for more features. The total size will be less now, but it requires that all the source files are recompiled before linking, otherwise the total size will be too large. Does it help if you force recompilation of all the sources?

beau-seidon, 2023-11-18:
I also don't get a compile error, in @Jaffe- latest commit, or @circuitcreature new commit with midi cc.

circuitcreature, 2023-11-18:
Im running gcc 13.2 on Mac.

Jaffe-, 2024-01-25:
Did you manage to resolve this by compiling all sources again? (delete all .o files)

circuitcreature, 2024-01-31:
I have not been able to get it to compile without changing it back to -O3. 
I thought it was fixed last night, but it's showing up again. I've run make clean and also 
`find ./ -type f \( -iname \*.o -o -iname \*.ilk -iname \*.map -o -iname \*.exp -iname \*.bin -o -iname \*.elf -o -iname \*.hex \) -delete`  just to make sure all comile files are gone.

beau-seidon, 2024-10-04:
@circuitcreature do you still get compile errors when you build with **make** and -O2 opt. without PlatformIO?
The current version compiles fine for me on Linux (RasPI OS - Buster [v10]) with no warnings or errors.

circuitcreature, 2024-10-04:
@beau-seidon I do. I can dig into this later. I might be able to dig up an older Mac to test on as well. Same error as before
`sample/sample.o: in function `.L22':
sample.c:(.text.allocate_block+0x176): relocation truncated to fit: R_AVR_7_PCREL against `.L9'
collect2: error: ld returned 1 exit status
make: *** [nesizer_fw.elf] Error 1
`

beau-seidon, 2024-10-04:
Ok, I'll dig into that as well to see what that function specifically does. Like, way down in there.
It may be related to the issues I'm having on my pitched-sample-playback branch! (I doubt it. I'm probably just doing something wrong)

My brother has a Mac now, I'll ask him to try and compile it on his lappy too to see if he gets the same error.
I have never been able to get make to work on Windows, but I'll fiddle around with it.

beau-seidon, 2024-10-04:
https://stackoverflow.com/questions/8188849/avr-linker-error-relocation-truncated-to-fit 
also this seems relevant

circuitcreature, 2024-10-08:
Going to close this issue. 
After digging in the code and trying umpteen things with Makefile, I purged avr-gcc and reinstalled it using a different version(14) than the default 9,  and It's compiling just fine now.

circuitcreature, 2024-10-08:
Closed. avr-gcc was out of date.  
For anyone else that ends up here, MacOS terminal:
`brew remove avr-gcc`
`brew install avr-gcc@14`

closed 2024-10-08 by circuitcreature