Jaffe-/NESizer2 #24: Setting the right fuses

NilocEidlog, 2024-05-17:
So .. after building this thing and programming it .. I turned it on and nothing happened. Lots of messing around ... anyway worked out how to set the fuses - tried doing a PR on the make file but I dont have permissions :(

Here's a nice little patch that allows you to do "make fuse" and it will burn the fuses correctly?? - pulled the values out of platformio.ini in this repo - all standard "arduino" fuse settings and any other avr stuff ive done in the past didnt work.

+fuse:
+       avrdude -u -c $(PROGRAMMER) -P usb -p $(MCU) -U efuse:w:0xFF:m -U lfuse:w:0xE0:m -U hfuse:w:0xD9:m
+

beau-seidon, 2024-05-20:
If you open the project in VSCode with the PlatformIO extension installed (on any OS), all you have to do is open the PlatformIO Tasks menu and select Set Fuses and it will burn the fuse bits as configured in platformio.ini.

[image]

Otherwise, if you have avrdude installed (it comes with the Arduino IDE), you can use it just as you wrote.
The specific command I used to burn the fuse bits, back when I compiled and loaded everything using the terminal (before I found out how to do it with PIO), was this:
```
avrdude -C "C:\Program Files (x86)\Arduino\hardware\tools\avr\etc\avrdude.conf" -v -p atmega328p -c stk500v1 -P COM4 -b 19200 -U lfuse:w:0xe0:m -U hfuse:w:0xd9:m -U efuse:w:0xff:m
```

And to flash the firmware I would use:
```
avrdude -C "C:\Program Files (x86)\Arduino\hardware\tools\avr\etc\avrdude.conf" -v -p atmega328p -c stk500v1 -P COM4 -b 19200 -U "flash:w:%USERPROFILE%\Desktop\nesizer_fw.hex:i"
```

The stk500v1 is the programmer argument that corresponds to an Arduino Uno with an ATmega328p installed which has been programmed with the "ArduinoISP.ino" example sketch. (I mention this primarily to inform anyone else who reads this in future who wants to use an Arduino Uno as a programmer.) Of course you will have to set your own paths and COM port settings etc, corresponding to your OS.

For a little more information, see the closed Issue called AVR Programming Method #11

NilocEidlog, 2024-05-25:
Cool - makes sense! I was just doing it all via cmd line in bash :P

closed 2024-05-25 by NilocEidlog

beau-seidon, 2024-05-25:
> Cool - makes sense! I was just doing it all via cmd line in bash :P

That's how it was originally intended to be done, but it's open source, build it however you want!

I still use Windows primarily. I love Linux but there are 2 applications I won't go without, which don't run properly on Linux. I was able to set fuse bits and load the firmware via Windows, but (probably because of operator error) I was only able to get the source to compile on Linux. Tried using git bash on Windows and mingw64, but the main problem is I still don't fully understand Make. Feel free to ridicule me.

I've been programming microcontrollers using PlatformIO for a while now, so I fiddled around a bit and got the NESizer2 firmware to compile and upload, on Windows and Linux, entirely within VSCode. I was told it works on MacOS also. 

All you have to do is git-clone or manually download the repo, open it in VSCode (with PIO extension enabled), set the fuse bits (only required the first time), and click upload. It will auto install any dependencies, compile everything, and load it to the chip using the programmer settings in the platformio.ini file, which may have to be modified to match your ISP programmer device. 

That's the way I like doing it.

beau-seidon, 2024-05-25:
If I remember correctly, @Jaffe- compiles (or has compiled at least once) via CLI using WSL in Windows. I haven't tried that yet.