Jaffe-/NESizer2 #29: MIDI causing NESizer to crash.

jeff464, 2024-08-16:
Exactly as the title says. When I hit play on my MPC Live, the NESizer blanks out. I can play notes all day long just fine, but as soon as I hit play, it's toast. Also worth noting, I can duplicate this behavior if I have my Roland MC-505 connected via MIDI and press any of the track mute buttons, but play and notes seem to send to the NESizer just fine. Also, Ableton notes/play work fine as well. I can duplicate the behavior on both of my NESizer units.

beau-seidon, 2024-08-21:
if you have a usb midi interface, hook up the midi output of your mpc live to the interface input and use midi-ox to capture the message it sends when you press play. that will help with diagnosis. 

try to capture any other controllers' midi messages that you have seen crash nesizer as well.

jeff464, 2024-09-15:
So it would seem as though the MPC no longer crashes the NESizer, but I think this *might* be due to a recent firmware update from Akai. However, I noticed that powering the Roland MC-505 caused the NESizer to crash if MIDI was connected. Here's what MIDIOX returned:

[image]

jeff464, 2024-09-15:
Here is output from the Roland MC-505 when I press a track button - it sends a sysex message, and this is ultimately what causes the NESizer to crash.

[image]

beau-seidon, 2024-09-15:
I'll see if I can figure out how it handles sysex and make it ignore sysex messages not specifically meant for NESizer.

beau-seidon, 2024-09-15:
I have added a statement which checks for the proper SysexID and DeviceID in the sysex message handler, and it seems to functioning correctly. 

However, I had to also edit the program which converts samples and sequences to .syx files, and I'm still testing that to make sure it's working properly and samples get uploaded to the right slots and play back etc. 

Will upload to a new branch on my fork for public testing when I get it working.

beau-seidon, 2024-09-15:
https://github.com/beau-seidon/NESizer2/tree/sysex-filter

Working fine now on my end. 
I included the recompiled gensysex.exe file for converting samples to sysex with the ID's in the message header. 
Let me know if you have issues with sample loading, playback, or more hangs / midi-handling misbehavior.

One thing I noticed, is that on infrequent occasion, it seems to ignore a correctly addressed sysex message from MIDIOX, and I have to send it again. Doesn't happen often, and when I resend, it usually works the second time.

jeff464, 2024-09-15:
Great - just uploaded, and the crashes seem to be fixed! That said, is there a trick to getting the MIDI CCs to work? It seems I'm not able to adjust anything via MIDI.

jeff464, 2024-09-15:
Wait, sorry - just realizing these are two different forks. So I assume it's now a choice between crashes or MIDI CC?

beau-seidon, 2024-09-15:
Ruh roh. Give me a little bit, I'm busy at the moment, but I will make the sysex changes to a clone of circuitcreature's fork and upload so you can play with it. It should be a pretty straightforward copy/paste process now. 

As far as the MIDI CC features, I haven't tested their fork very much on my own synth yet so I don't know what issues to expect. Let me know how it goes!

beau-seidon, 2024-09-16:
> Wait, sorry - just realizing these are two different forks. So I assume it's now a choice between crashes or MIDI CC?

https://github.com/beau-seidon/NESizer2/tree/circuitcreature-fix-sysex

beau-seidon, 2024-09-21:
Adding the ignore function based on SysEx messages' "ID bytes" fixed the crashes from irrelevant sysex messages, but I found that sometimes the synth is missing bytes (race condition?) which can result in switching to the IGNORE state erroneously.

I'm working on a branch that has already fixed that issue but I still have a couple more things to test. Might have it committed this evening.

I also added patch selection via SysEx which works just like ProgramChange.

closed 2024-10-04 by beau-seidon