Jaffe-/NESizer2 #30: Loud MIDI zipper noises/note clicking when sending MIDI to NESizer

jeff464, 2024-08-16:
I am getting pretty bad zippering/clicking whenever either of my NESizer units receive any sort of MIDI data. If the channels are not engaged, it's loud enough to register on my MOTU audio interface/oscilloscope. I don't seem to get this on any of my other synths or Famimimidi. I tested different power supplies, outlets, and cables, none of which had any change.

Any idea what might be happening here? It's making it difficult to make softer/quieter sounds on the NESizer.

Jaffe-, 2024-10-02:
@jeff464: Unfortunately there is a lot of noise bleed on the NESizer, it is just the way it is - poorly designed in this regard :p. Remember it has strictly been a hobby project by a (at the time) electronics engineer student and never designed with the intention of selling as a product. 

Someone more knowledgeable in electronics or with audio knowledge might try and come up with some fixes for existing boards, but I think the PCB has to be redesigned with proper separation between audio and digital signal and power domains to get rid of this. Maybe someone comes up with something clever (or even better, a NESizer v3!)

beau-seidon, 2024-10-03:
That sound is called "DIY Character" :)

If we make a NESizer3, let's use an ATmega1284 because it has extra UARTs!