Jaffe-/NESizer2 #31: MIDI CCs not applying to related MIDI channels when voices on NESizer are stacked.

jeff464, 2024-09-17:
As title says - If the user has SQ1/SQ2/TRI set to the same MIDI channel on the NESizer and tries adjusting a MIDI CC parameter they all share (MIDI CC5 for glide, for example) it only affects SQ1 instead of applying to all related voices. 

It's definitely not the end of the world, but it would be nice to have MIDI CCs apply to everything universally for instances when voices are stacked on the NESizer side.

beau-seidon, 2024-09-17:
Keep in mind that this main fork does not yet officially include MIDI CC support. 
I think it would be more proper to move this issue to @circuitcreature 's [fork](https://github.com/circuitcreature/NESizer2), but he would need to enable the Issues tab from his Settings page.

Then again it might be better to keep all discussion in one place?
I suppose it would be up to @Jaffe- since you're writing on his wall. lol

jeff464, 2024-09-17:
Yeah, I considered writing on the forked wall, but noticed the lack of issues tab. I also figured since most activity is happening here, what's a little more? :)